LacunaSpace/basicmac #4: Should verify that AVR and/or SX1272 still work

matthijskooijman, 2020-04-15:
This code is intended to work on AVR and I tested an early version of the Arduino port successfully on an AVR board with an SX1272 radio chip, but the latest master does not seem to work anymore on that board (tried the abp example, serial output looks fine and runs to TX_COMPLETE fine, but no data comes in on the gateway).

It does work properly on and STM32 and SAMD boards, with SX1276 and SX1262, so I suspect that either AVR or SX1272 has broken since my early tests, but I'm not sure which.

I also tried to `prototype` branch on AVR, which I had expected to still work on that board, which also did not work, so maybe this has been broken longer than I thought, or maybe my hardware is broken...

finrut, 2020-04-20:
Hello Matthijs,

I tried the `ttn-otaa` example but I've got this error :

```
Starting

SPI config
MOSI : 23
MISO : 19
SCK  : 18
SS   : 5

00:00:00.040: /home/github/basicmac/src/lmic/radio-sx127x.c:1122: assertion failed
```

At the moment I don't have an Arduino or STM32 capable for the lib (on UNO it doesn't fit/run),
so I'm using an ESP32 running Arduino frmwk plus a LoRa RFM95 module (SX1276 chip);
pins assignment is the usual one I also used with your great previous lib `Arduino-LMIC`.

I'm a bit confused why is giving me that assert (SPI pins seem correct too), 
can you help me?

matthijskooijman, 2020-04-20:
Thanks for the feedback. I've moved it to #10 to keep this issue a bit concise.

matthijskooijman, 2020-05-04:
As for AVR and SX1272, both still seem to work with the current master branch. There does seem to be a problem with output power, which is why it seemed to not work before, but that same problem happens with LMIC, so I suspect this is either a hardware problem (maybe I blew up the output stage at some point), or maybe this radio board has a funny TXRX or PABOOST configuration that neither BasicMAC and LMIC correctly configure (in which case I just haven't noticed this before).

For future reference, I tested the ABP test sketch on my SX1272 board, which resulted in around -60 to -80 dB RSSI and around 10 dB SNR on just my own gateway which is in the same room. I got the same results with LMIC and BasicMAC. Then, I tested the same sketch on a SX1262 board, with the same (whip) antenna, and got around -20 dB RSSI and again around 10 dB SNR on the same gateway (plus reception on half a dozen gateways.

I'll leave this issue open for a bit, to remind me to further investigate the output stage settings for this board.