LacunaSpace/basicmac #8: Should verify whether non-EU868 regions work

matthijskooijman, 2020-04-16:
So far, I've only tested EU868. In #5, some issues with US915 were reported, but maybe the 2.2 update that occurred after that issue fixes these issues?

Anyway, this just needs some testing. If you have hardware for other regions and want to give this a spin, please report your findings here :-)

ricaun, 2020-04-16:
Hello,
I gonna try mix some code from version 2.2, to try to fix the join.

I noticed de the library does not have a good way to select a subband
like `LMIC_selectSubBand`.

I created this dump function to do the job. 

```
void LMIC_selectSubBand(int sub)
{
  if (REG_IS_FIX()) {
    for (int i = 0; i < MAX_FIX_CHNLS_125; i++)
    {
      if ((sub * 8 - 1) < i && i < (sub + 1) * 8)
      {
        Serial.print("LMIC_enableChannel : ");
        Serial.println(i);
      }
      else
      {
        LMIC_disableChannel(i);
      }
    }

    for (int i = 0; i < MAX_FIX_CHNLS_500; i++)
    {
      if (sub == i)
      {
        Serial.print("LMIC_enableChannel : ");
        Serial.println(MAX_FIX_CHNLS_125 + i);
      }
      else
      {
        LMIC_disableChannel(MAX_FIX_CHNLS_125 + i);
      }
    }
  }
}
```

It's not pretty but do the job.

matthijskooijman, 2020-07-15:
Another thing to test is whether enabling multiple regions and selecting one at runtime actually works.

@ricaun With your suggested `LMIC_selectSubBand()`, were you able to test US915? If that works, I'm inclined to just reinstate that function (using your suggested version, or maybe something slightly more internal).

ricaun, 2020-07-15:
> @ricaun With your suggested `LMIC_selectSubBand()`, were you able to test US915? If that works, I'm inclined to just reinstate that function (using your suggested version, or maybe something slightly more internal).

Yes, I have US915 and AU915 gateways in my lab, I already test each region. Works fine! (Downlink is not working, probably TTN server is buggy).

I didn't test both regions at the same time, I need to define both regions on the target-config and to select the region on the code using some like this???

```
u1_t code = 0; // 0 or 1
u1_t os_getRegion (void) { return LMIC_regionCode(code); }
```
That's it? If I change the `code` I can select the first (code = 0;) or second (code = 1;) region? 

I supposed that's the way to select the region on the code...

matthijskooijman, 2020-07-15:
> That's it? If I change the code I can select the first (code = 0;) or second (code = 1;) region?

Yup, I think that should be enough. You can also just return e.g. `REGCODE_US915` from `os_getRegion()`, then you don't have to rely on the order of regions (but for a quick test, either is fine).