LacunaSpace/basicmac #10: radio-sx127x.c:1122 assertion failed on ESP

matthijskooijman, 2020-04-20:
Hello Matthijs,

I tried the `ttn-otaa` example but I've got this error :

```
Starting

SPI config
MOSI : 23
MISO : 19
SCK  : 18
SS   : 5

00:00:00.040: /home/github/basicmac/src/lmic/radio-sx127x.c:1122: assertion failed
```

At the moment I don't have an Arduino or STM32 capable for the lib (on UNO it doesn't fit/run),
so I'm using an ESP32 running Arduino frmwk plus a LoRa RFM95 module (SX1276 chip);
pins assignment is the usual one I also used with your great previous lib `Arduino-LMIC`.

I'm a bit confused why is giving me that assert (SPI pins seem correct too), 
can you help me?

_Originally posted by @finrut in https://github.com/LacunaSpace/basicmac/issues/4#issuecomment-616400662_

matthijskooijman, 2020-04-20:
That assertion is this one:

> https://github.com/LacunaSpace/basicmac/blob/4c399a2ca0f07bdc8d1e68145980ee494eb92e60/lmic/radio-sx127x.c#L1122

Do you have a reset pin in your pin map? This might have been added since BasicMAC, and if you do not define a reset pin, you *must* set it to LMIC_UNUSED_PIN so the library can act accordingly.

matthijskooijman, 2020-04-20:
Be sure to check all of your pinmap and add any missing fields, since they default to 0 which refers to pin 0, not unused pin...

finrut, 2020-04-21:
Hello Matthijs,

first of all thank you for your replies,

today I've checked again the configuration and now it's working like a charm
(maybe I've missed some pins as I was using platformio project).

```
Starting
SPI pins
MOSI : 27
MISO : 19
SCK : 5
SS : 18

lmic pins
ss   : 18
rst  : 14
dio0 : 26
dio1 : 33
dio2 : 32
tx   : 255
rx   : 255
busy : 255
tcxo : 255

00:00:05.082: engineUpdate[opmode=0x8]
Packet queued # : 0
318380: EV_JOINING
...
...
1251020: EV_JOINED
...
...
00:00:21.142: engineUpdate[opmode=0x888]
00:00:21.142: Received downlink[window=RX1,port=-1,ack=0]
1321404: 
EV_TXCOMPLETE
00:00:21.145: engineUpdate[opmode=0x800]
00:00:21.149: engineUpdate[opmode=0x808]
Packet queued # : 1
...
```

I have another couple of questions but not related to this (can be marked closed),
so I'll open a separated thread.

Thanks

matthijskooijman, 2020-04-21:
Thanks for reporting back, closing. Feel free to open more issues with additional questions :-)

closed 2020-04-21 by matthijskooijman