LacunaSpace/basicmac #11: Questions about class C and down windows

finrut, 2020-04-21:
Hi Matthijs,

I've seen downlinks are fine (in original LMIC-Arduino I couldn't receive pkts > 51 bytes)
and I'm trying the Class C (I haven't found documentation or example on it)
and I have 2 question :
- how to use LMIC_setClassC func? (I would change from Class A to C and back)
- is it possible to put a (class C) node in listening to receive downlinks without send any uplink?

for the second one questione I was thinking something like this from the raw.ino example:
```
void rx(osjobcb_t func) {
  LMIC.osjob.func = func;
  LMIC.rxtime = os_getTime(); // RX _now_
  // Enable "continuous" RX (e.g. without a timeout, still stops after
  // receiving a packet)
  os_radio(RADIO_RXON);
  Serial.println("RX");
}

static void rx_func (osjob_t* job) {
  // next RX
  os_setCallback(&rxjob, rx_func);

if (LMIC.dataLen != 0 )
{
  Serial.print("Got ");
  Serial.print(LMIC.dataLen);
  Serial.println(" bytes");
  Serial.write(LMIC.frame, LMIC.dataLen);
  Serial.println();
}
  // Restart RX
  rx(rx_func);
}
```

but I'm not sure because the dutycycle in LoRaWAN, so I'm wondering if I can simple RX directly through Basicmac (or LMIC ) API.

Do you have any suggestions?

matthijskooijman, 2020-04-21:
Hm, I've seen some references to class C, but TBH I have not looked at how this works at all (or even if it still works, especially when you disable class B there is a chance some things needed for class C are also disabled).

> is it possible to put a (class C) node in listening to receive downlinks without send any uplink?

Isn't that exactly what class C means?

As for your example, I suspect that simply enabling `RXON` might not be the way to go, since then the library likely does not have the right stuff setup to handle the RX (though you're sortof doing that manually, but I think BasicMAC might be able to do this for you).

Also, you speak of duty cycle, but there is no duty cycle for RX, so that should be ok?

finrut, 2020-04-21:
Hello Matthijs,

sorry from the readme I had intended this lib could handle class C devices,
anyway are you planning tests for this feature?

> is it possible to put a (class C) node in listening to receive downlinks without send any uplink?
> 
> Isn't that exactly what class C means?

practically yes ( a "full" class C node could also send uplinks AFAIK),
in my case I don't need to TX during C session.

About the duty-cycle, I was not referring to lorawan limitations, 
you are right duty-cycle only applies to end-device transmissions,
but I meant the LMIC scheduling of RX windows, as it uses a great system of scheduled callbacks to match the exact rxtime windows,
but for my case I was thinking to use the RXCONTINUOS mode from the radio;
so, leaving out RXON opmode, which are the API/funcs from BasicMac you were suggesting?

matthijskooijman, 2020-04-21:
> which are the API/funcs from BasicMac you were suggesting?

Well, you suggested there is an `LMIC_setClassC` function, so that one? :-)

finrut, 2020-04-21:
Ok, I'll give a try with the LMIC_setClassC and let you know.
Thank you for your time :)

finrut, 2020-04-28:
Hi Matthijs,

I've tried calling the Class C API 

```
in_class_c_mode = true;
if(LMIC.clmode != CLASS_C)
    {
        printf("setting LMIC class C\n");
        LMIC_setClassC(in_class_c_mode);
    }
    else
        printf("already in class C\n");

    if(LMIC.clmode != CLASS_C)
        printf("still not LMIC class C\n");
    else
        printf("now LMIC class C\n");
```

and from logs it seems correctly set, but my node doesn't receive data from the server (I checked it uses same channel, sf and bw)  :

```
setting LMIC class C
00:03:22.670: engineUpdate, opmode : 800
00:03:22.674: RXON_MODE[freq=868.1,sf=7,bw=125]
now LMIC class C
E (215210) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (215210) task_wdt:  - IDLE1 (CPU 1)
E (215210) task_wdt: Tasks currently running:
E (215210) task_wdt: CPU 0: IDLE0
E (215210) task_wdt: CPU 1: loopTask
E (220210) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (220210) task_wdt:  - IDLE1 (CPU 1)
E (220210) task_wdt: Tasks currently running:
E (220210) task_wdt: CPU 0: IDLE0
E (220210) task_wdt: CPU 1: loopTask
...
...
```

If I try to call a fake job (also from onLmicEvent EV_RXCOMPLETE) :
```
in_class_c_mode = true;
if(LMIC.clmode != CLASS_C)
        LMIC_setClassC(in_class_c_mode);
os_setCallback(&alive_job, alive_cb);
```

where
```
void alive_cb(osjob_t *j)
{
        printf("alive_cb, nop\n");
}
```

Class C reception phase starts :

```
setting LMIC class C
00:03:22.670: engineUpdate, opmode : 800
00:03:22.674: RXON_MODE[freq=868.1,sf=7,bw=125]
now LMIC class C
00:03:22.679: Scheduled job 1073426608, cb 1074614084 at 00:03:23.879
00:03:23.879: Running job 1073426608, cb 1074614084, deadline 00:03:23.879
alive_cb, nop
00:03:25.608: Scheduled job 1073435356, cb 1074830744 ASAP
00:03:25.608: Running job 1073435356, cb 1074830744, deadline 00:03:25.608
00:03:25.610: irq_proc : RX[freq=868.1,sf=7,bw=125,rssi=-86,snr=7.50,len=255]
00:03:25.616: Scheduled job 1073434184, cb 1074826152 ASAP
00:03:25.621: Running job 1073434184, cb 1074826152, deadline 00:03:25.616
00:03:25.631: Received downlink[window=RX2,port=1,ack=0]
12852089: EV_RXCOMPLETE
I (213331) rx: port : 1
I (213331) rx: len : 242
data # 1

00:03:25.643: Scheduled job 1073426608, cb 1074614084 ASAP
00:03:25.648: engineUpdate, opmode : 800
00:03:25.652: RXON_MODE[freq=868.1,sf=7,bw=125]
00:03:25.656: Running job 1073426608, cb 1074614084, deadline 00:03:25.643
alive_cb, nop
E (215211) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (215211) task_wdt:  - IDLE1 (CPU 1)
E (215211) task_wdt: Tasks currently running:
E (215211) task_wdt: CPU 0: IDLE0
E (215211) task_wdt: CPU 1: loopTask
00:03:27.626: Scheduled job 1073435356, cb 1074830744 ASAP
00:03:27.626: Running job 1073435356, cb 1074830744, deadline 00:03:27.626
00:03:27.627: irq_proc : RX[freq=868.1,sf=7,bw=125,rssi=-86,snr=7.25,len=255]
00:03:27.634: Scheduled job 1073434184, cb 1074826152 ASAP
00:03:27.639: Running job 1073434184, cb 1074826152, deadline 00:03:27.634
00:03:27.649: Received downlink[window=RX2,port=1,ack=0]
12978189: EV_RXCOMPLETE
I (215341) rx: port : 1
I (215351) rx: len : 242
data # 2
```

but not for every try.

Do you have any advice?

matthijskooijman, 2020-04-28:
Weird, so scheduling a job somehow causes reception to work? That's quite weird, especially since the IRQ does not even seem to trigger in the non-working case... No specific advance off-hand, sorry.

finrut, 2020-04-28:
> Weird, so scheduling a job somehow causes reception to work? 

Yes it looks like so, but I'm not sure if the node is receiving in RXON mode or, maybe,
in RX SINGLE thanks to the engine running.. (I had a look to the Basicmac code but I couldn't go further)

Anyway thanks for your attention, I'll update in case.

cdrx, 2022-01-25:
@finrut were you able to get class C working?

finrut, 2022-01-27:
@cdrx Currently I'm not working on it,
in the last tests I made at that time (April 2020) I had no luck with this or other lmic-based libraries,
just the mbed-os fuota demo gave me partially success results, but not so reliable due to loraserver (chirpstack) instabilities at that time.
I've seen there is a new separated version of fuota server, so maybe you can have a chance from there.