LacunaSpace/basicmac #12: Devnonce issues raised from network server

finrut, 2020-04-28:
Hi Matthijs,

I've incurred this issue during my tests with BasicMac on ESP32 :
```
time="2020-04-28T10:05:02Z" level=info msg="uplink: frame(s) collected" ctx_id=e401a203-720f-43e0-a8bc-909c818ca0a4 mtype=JoinRequest uplink_ids="[cd957e54-084d-47b7-87a5-69a6a7de92ab]"
time="2020-04-28T10:05:02Z" level=error msg="uplink: processing uplink frame error" ctx_id=e401a203-720f-43e0-a8bc-909c818ca0a4 error="validate dev-nonce error: object already exists"
```

Do you know what it can be?

matthijskooijman, 2020-04-28:
I think the dev nonce is now just randomly selected (rather than incrementally, as LoRaWAN 1.1 specifies IIRC, since that requires storing the last used nonce in EEPROM or similar). However, it might very well be that the random generator is never seeded, so you'll get the same nonce on every try.

You could fix that by something like `randomSeed(analogRead(A0));` (e.g. use a random reading from an unused analog pin as the seed, but that might not be sufficient entropy (at most 1024 different values, in practice likely fewer). Seeding using some radio signal measurement might be more reliable.

The SX1262 has an internal random generator too, and there is code for that, but it always returns 0 for me, so I'm not sure what's up with that one.

I had thought of this before and wanted to check this, but since things just worked, I haven't gotten around to it I think.

finrut, 2020-04-28:
> The SX1262 has an internal random generator too

Bad luck for me, I'm testing on a node with old SX1276 module.

I haven't said before (maybe it can be notable), the problem appears after some resets of the node,
and I registered the node on the server as LoRaWAN 1.0.2 version.

So I'll try to use the random seed (or radio sig as you said).

Again, thanks a lot for your support.