LacunaSpace/basicmac #26: Atmega328 1MHz

Samuel-ZDM, 2020-11-22:
Hello!

I'm using an Atmega328 with 1MHz with RFM95, but I can't send a message to the gateway.
My device sends the first message and then no longer sends it.
I am using the ABP example to send to TTN.

Can you help me?

matthijskooijman, 2020-11-22:
BasicMAC is not quite optimized for AVR and will probably eat up pretty much all of your flash on a 328p (leaving no flash for a meaningful application or even debug output, IIRC), so I'm not really planning on supporting the 328p, since it's just not really feasible.

If you're using an RFM95, which has an SX127x chip, I recommend you just use https://github.com/mcci-catena/arduino-lmic instead.

I'll go ahead and close this issue, but feel free to leave followup comments or (ask to) reopen if needed.

closed 2020-11-22 by matthijskooijman

Samuel-ZDM, 2020-11-23:
I'm using this repository, I just thought that the issues there were not being answered. So I'm going to open that same issue in the lmic repository.

Thanks!

Samuel-ZDM, 2020-11-23:
I sent there for support but I didn't get a return. That's why I put it here.

Thanks!

matthijskooijman, 2020-11-23:
Do you realize that LMIC and BasicMAC are related, but significantly different libraries? If you're using one, it makes no sense to report an issue against the other... If you're using the MCCI LMIC version, than that other repo is the right place for reporting issues. However, realize that issue trackers are often used for reporting bugs and requesting features, they do not always work as a general purpose way to get "support" (you might try the Arduino forum, the TTN forum, or some chat channels for that maybe). Also, your post is quite low on information, a bit more info on what you've tried already, or maybe some debug output could be helpful to add when you ask this question (but don't bother adding it here, as this is the wrong place).

Samuel-ZDM, 2020-11-23:
Yes, sorry about that.
I thought that the same principle used in the lmic of your repository was used in the other, because my question is whether the lower frequency of the micro can influence the functioning of the communication. I thought it was something more general and so you could help me.
But sorry, I will add more information in the other issue.

Thanks!