LacunaSpace/basicmac #32: not receiving rx

wemaxon, 2021-09-13:
Hello,

I am using the BasicMAC library with Arduino IDE on an [Sparkfun LoRa Thing Plus](https://www.sparkfun.com/products/17506). The Board features an NM180100 Module with an apollo3 microcontroller and sx1262 transceiver. I have set up my own chirpstack lora serverstack with a mikrotik gateway.

I am opening this issue because I am having problems with the Rx part. The module wont receive any frames sent from the gateway. The OTAA activation also takes lots of attempts until established. The Tx messages on the other hand do work, once the join is successfull.

The following is the serial debugging output when the device is trying to join and and then transmit:
```
[MAIN]Application started<\r><\n>
[lora]Initializing LMIC<\r><\n>
[lora]Joining<\r><\n>
[lora]EV_JOINING<\r><\n>
00:00:00.055: engineUpdate[opmode=0x4]<\r><\n>
00:00:06.000: engineUpdate[opmode=0x4]<\r><\n>
[lora]EV_TXSTART<\r><\n>
00:00:06.006: engineUpdate[opmode=0x884]<\r><\n>
00:00:06.010: TX[mod=LoRa,sf=7,bw=125,cr=4/5,nocrc=0,ih=0,fcnt=0,freq=868.1,pow=16,len=23]: 000000000000000000AEB4F4B34759576F0FBDD1E1E628<\r><\n>
[lora]EV_TXDONE<\r><\n>
00:00:06.093: engineUpdate[opmode=0x884]<\r><\n>
00:00:11.086: RX_MODE[mod=LoRa,sf=7,bw=125,cr=4/5,nocrc=1,ih=0,freq=868.1,rxtime=693245]<\r><\n>
00:00:11.100: WARNING: rxtime is 530 ticks in the past! (ramp-up time 5 ms / 365 ticks)<\r><\n>
00:00:11.114: RX: TIMEOUT<\r><\n>
00:00:12.102: RX_MODE[mod=LoRa,sf=12,bw=125,cr=4/5,nocrc=1,ih=0,freq=869.5,rxtime=756737]<\r><\n>
00:00:12.116: WARNING: rxtime is 537 ticks in the past! (ramp-up time 5 ms / 364 ticks)<\r><\n>
00:00:12.321: RX: TIMEOUT<\r><\n>
00:00:12.442: engineUpdate[opmode=0x4]<\r><\n>
[lora]EV_TXSTART<\r><\n>
00:00:12.448: engineUpdate[opmode=0x884]<\r><\n>
00:00:12.452: TX[mod=LoRa,sf=7,bw=125,cr=4/5,nocrc=0,ih=0,fcnt=0,freq=868.3,pow=16,len=23]: 000000000000000000AEB4F4B34759576FCD3CF5516130<\r><\n>
[lora]EV_TXDONE<\r><\n>
00:00:12.535: engineUpdate[opmode=0x884]<\r><\n>
00:00:17.528: RX_MODE[mod=LoRa,sf=7,bw=125,cr=4/5,nocrc=1,ih=0,freq=868.3,rxtime=1095869]<\r><\n>
00:00:17.542: WARNING: rxtime is 535 ticks in the past! (ramp-up time 5 ms / 364 ticks)<\r><\n>
00:00:17.556: RX: TIMEOUT<\r><\n>
00:00:18.544: RX_MODE[mod=LoRa,sf=12,bw=125,cr=4/5,nocrc=1,ih=0,freq=869.5,rxtime=1159361]<\r><\n>
00:00:18.558: WARNING: rxtime is 541 ticks in the past! (ramp-up time 5 ms / 364 ticks)<\r><\n>
00:00:18.763: RX: TIMEOUT<\r><\n>
00:00:29.618: engineUpdate[opmode=0x4]<\r><\n>
[lora]EV_TXSTART<\r><\n>
00:00:29.623: engineUpdate[opmode=0x884]<\r><\n>
00:00:29.627: TX[mod=LoRa,sf=8,bw=125,cr=4/5,nocrc=0,ih=0,fcnt=0,freq=868.5,pow=16,len=23]: 000000000000000000AEB4F4B34759576F9789CFD63655<\r><\n>
[lora]EV_TXDONE<\r><\n>
00:00:29.762: engineUpdate[opmode=0x884]<\r><\n>
00:00:34.755: RX_MODE[mod=LoRa,sf=8,bw=125,cr=4/5,nocrc=1,ih=0,freq=868.5,rxtime=2172563]<\r><\n>
00:00:34.769: WARNING: rxtime is 538 ticks in the past! (ramp-up time 5 ms / 367 ticks)<\r><\n>
00:00:34.790: RX: TIMEOUT<\r><\n>
00:00:35.771: RX_MODE[mod=LoRa,sf=12,bw=125,cr=4/5,nocrc=1,ih=0,freq=869.5,rxtime=2236023]<\r><\n>
00:00:35.785: WARNING: rxtime is 539 ticks in the past! (ramp-up time 5 ms / 363 ticks)<\r><\n>
00:00:35.989: RX: TIMEOUT<\r><\n>
00:00:47.004: engineUpdate[opmode=0x4]<\r><\n>
[lora]EV_TXSTART<\r><\n>
00:00:47.010: engineUpdate[opmode=0x884]<\r><\n>
00:00:47.013: TX[mod=LoRa,sf=8,bw=125,cr=4/5,nocrc=0,ih=0,fcnt=0,freq=868.1,pow=16,len=23]: 000000000000000000AEB4F4B34759576FB6694AF9FDBF<\r><\n>
[lora]EV_TXDONE<\r><\n>
00:00:47.149: engineUpdate[opmode=0x884]<\r><\n>
00:00:52.142: RX_MODE[mod=LoRa,sf=8,bw=125,cr=4/5,nocrc=1,ih=0,freq=868.1,rxtime=3259225]<\r><\n>
00:00:52.156: WARNING: rxtime is 536 ticks in the past! (ramp-up time 5 ms / 364 ticks)<\r><\n>
00:00:52.176: RX: TIMEOUT<\r><\n>
00:00:53.157: RX_MODE[mod=LoRa,sf=12,bw=125,cr=4/5,nocrc=1,ih=0,freq=869.5,rxtime=3322685]<\r><\n>
00:00:53.171: WARNING: rxtime is 542 ticks in the past! (ramp-up time 5 ms / 365 ticks)<\r><\n>
00:00:53.376: RX: TIMEOUT<\r><\n>
00:01:18.377: engineUpdate[opmode=0x4]<\r><\n>
[lora]EV_TXSTART<\r><\n>
00:01:18.382: engineUpdate[opmode=0x884]<\r><\n>
00:01:18.386: TX[mod=LoRa,sf=9,bw=125,cr=4/5,nocrc=0,ih=0,fcnt=0,freq=868.3,pow=16,len=23]: 000000000000000000AEB4F4B34759576F02BFC6F3751F<\r><\n>
[lora]EV_TXDONE<\r><\n>
00:01:18.614: engineUpdate[opmode=0x884]<\r><\n>
00:01:23.608: RX_MODE[mod=LoRa,sf=9,bw=125,cr=4/5,nocrc=1,ih=0,freq=868.3,rxtime=5225870]<\r><\n>
00:01:23.622: WARNING: rxtime is 536 ticks in the past! (ramp-up time 5 ms / 364 ticks)<\r><\n>
00:01:23.655: RX: TIMEOUT<\r><\n>
[... 100 lines cut ...]
    irqlevel++;
}

void hal_enableIRQs () {
    if(--irqlevel == 0) {
        //interrupts();                  --> COMMENTED IN THE SPARKFUN MODIFIED BASICMAC LIBRARY

        // Instead of using proper interrupts (which are a bit tricky
        // and/or not available on all pins on AVR), just poll the pin
        // values. Since os_runloop disables and re-enables interrupts,
        // putting this here makes sure we check at least once every
        // loop.
        //
        // As an additional bonus, this prevents the can of worms that
        // we would otherwise get for running SPI transfers inside ISRs
        hal_io_check();
    }
}
...
```

Could this be about timing or because of the missing interrupt functionality on my board?

I'd be grateful for any help.
thanks in advance!

tfeseker, 2023-05-18:
Hi,

I am trying to solve the same problem... Has anyone ever managed to get the [Sparkfun LoRa Thing Plus](https://www.sparkfun.com/products/17506) working in OTAA mode?

I would appreciate if someone could point out a solution. Thanks!

wemaxon, 2023-05-19:
Hi @tfeseker,

i remember getting it to work. I think the Arduino IDE Code Examples provided by Sparkfun didnt have the correct setup for the  GPIO Connection to the Antenna RF Switch. Also I think I had to adjust a timing variable inside the library. 

I have attached my final working firmware including that includes the changes necessary to libs to get it to work. 
[lora-modul-firmware.zip](https://github.com/LacunaSpace/basicmac/files/11520306/lora-modul-firmware.zip)

If you have any questions let me know. Ill try to remember.